Chaoticlab/CNC-Tap-for-Voron #4: X-Limit Switch Bracket too Short for Carbon Fiber Gantry Kit

solrac8874, 2023-12-27:
I upgraded my Voron 2.4 with Chaoticlab's Carbon Fiber Gantry kit (A-B motor mounts + XY gantry mounts with 2020 extrusions) and CNC Tap V2.  The Carbon Fiber Gantry kit doesn't support X & Y limit switches on the XY gantry mounts so I relocated the Y-limit switch to the A motor mount per the Carbon Fiber Gantry kit manual.  Works great.

However, when relocating the X-limit switch to the CNC Tap V2 using the included Limit Switch Bracket I found the MGN12 Block hits the upper XY gantry mount plate before the X-limit switch hits the lower XY gantry mount plate/filler block.  Seems the Limit Switch Bracket is short by about 6-8mm to enable the X-limit switch to hit before the MGN12 Block.  Is there a fix for this?
See pictures below.

Front view:
[image]

My thumbnail is where the MGN12 Block hits the upper XY gantry mount plate.  The straight down the ruler shows the gap preventing the X-limit switch from making contact to the lower XY gantry mount plate/filler block.

Rear view:
[image]

With my Toolhead moved to max X-position limited by the MGN12 Block hitting the upper XY gantry mount plate, you can see the X-axis limit does not contact the lower XY gantry mount plate/filler block.

The V2 Bracket Adapter and the CNC_Voron_TAP_XY_Switch_Adapter_V2 do not seem to work or I don't see how those could be used for my case.  The adapters seem to be intended for use with the original Voron 2.4 XY-Limit Switch Mount configuration, not for use with Chaoticlab's Carbon Fiber Gantry kit.

I also noticed the original Voron 2.4 XY gantry mounts are flush top & bottom.  Whereas, the Chaoticlab's Carbon Fiber XY Gantry mounts are not; top plate matches the Voron XY gantry mount surface location, but the lower plate does not - source of the problem?

Chaoticlab, 2024-01-05:
Hey bud, thank you for commented this issues, we have design the MOD for help you better installed CNC Voron Tap V2 with carbon fiber kit, please check in this https://github.com/Chaoticlab/CNC-Tap-for-Voron/blob/master/STL/CNC_VORON_TAP_V2/V2%20Bracket%20Adapter.STL

solrac8874, 2024-01-06:
I tried using the CNC_Voron_TAP_XY_Switch_Adapter_V2, but it doesn't seem to work.  Am I installing it wrong?  I assume it screws into the right most (from the pic in my original post) hole of the Tap to extend the X-limit switch bracket to the right.  Unfortunately, I can't get to hole pattern in the Adapter to match up with the Tap right most hole.

Do you have graphic or picture of how the Adapter is installed to the Tap and X-limit switch bracket?

solrac8874, 2024-02-15:
I tried the limit switch bracket in pull request #6 , but it was still a hair short for me.  I modified the bracket design a bit by moving the switch mount holes an additional 2.5mm horizontally and 2mm vertically away from the bracket mount holes.  See attached pictures (orig. vs. mod).  May consider modifying the bracket in pull request #6 accordingly.

Original (from Pull Request #6 )
[image]

My mod make it work.
[image]

This issue can be closed once Pull Request#6 is closed.

Thanks for the support.