Chaoticlab/CNC-Tap-for-Voron #5: Stock X endstop doesn't get triggered

YoungHillionaire, 2023-12-30:


EddieGreen, 2024-02-08:
Not sure what you mean by stock endstop, but guessing you mean this:

[image]

The bolt hole does coincide with the PCB endstop microswitch - something I found out before I CADded the assembly.

I put the design on hold waiting for the V2 CAD model from Chaotic Labs, but in two months, not a word about that.

The V1 bolt positions and the important dimensions are the same between V1 and V2, so I have used the V1 assembly to make progress with the endstop design.

I thought I'd let you know that a design is on it's way - I'll post updates here.

YoungHillionaire, 2024-02-08:
Yes I need something like this to trigger the stock xy endstop

I have microswitches 
I didn't want to solder because my soldering is not working

YoungHillionaire, 2024-02-08:
I went to an electronics shop and get mine soldered so problem fixed for me

But they need to add an stl like this picture for people who want to use stock xy endstop

ruxi, 2024-03-22:
+1, also looking for x-endstop solution for ldo's xy pcb

EddieGreen, 2024-03-22:
I created a suitable X-endstop for Chaoticlabs Voron TAP V1-2 (Harkt XY Endstop PCB):

[image]

BOM: 3no. M3x8 SHCS, 1no. M3 Threaded Insert

[Chaoticlabs Voron TAP V1-2 (Harkt XY Endstop PCB) Trigger.zip](https://github.com/Chaoticlab/CNC-Tap-for-Voron/files/14723534/Chaoticlabs.Voron.TAP.V1-2.Harkt.XY.Endstop.PCB.Trigger.zip)

STEP file attached.

mabl, 2024-03-24:
I just used a long screw that I inserted into the bottom threads of the CNC-Tap. The screw head reliably hits and triggers the LDO XY end stop. I countered the screw with an additional nut.