Chaoticlab/CNC-Tap-for-Voron #9: Cable wiring for 2209

phpbbireland, 2024-03-08:
If you connect the connector to either of the EBB 2209 boards with orientation shown, V and G will be reversed possibly damaging the sensor board.

phpbbireland, 2024-03-10:
Correct wiring... 
[image]

Mugga6315, 2024-04-12:
I don't think that this is the correct orientation. I did it that way, light from tap is working and in klipper I get reading. But its always triggered when board klipper is started. Even when I switch in config. Seems that maybe G / S have to be switched. But I wanted to check first

hiddengerbil, 2025-02-22:
I have a premade cable, which fits that probe socket, and it has a tap plug and a two pin Filament run-out detector. Is the filament detector still possible with your wiring example?